bitsy-ai/octoprint-nanny-plugin #18: Unable to install plugin

arut16, 2020-12-27:
Hi,

I have the last version of Octoprint. Tried to install octo-nanny plugin. Takes around 5min then installation seems to be done but a notification shows install : "unknown" OK. After an octoprint reboot, plugin is not installed...
[image]
[image]
[image]
After install : 
[image]

INSTALLATION LOG :

> `Installation du plugin "OctoPrint Nanny" à partir de https://github.com/bitsy-ai/octoprint-nanny-plugin/archive/main.zip ...
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmp2fysfo75/octoprint-nanny-plugin-main.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp2fysfo75/octoprint-nanny-plugin-main.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (1.5.2)
Collecting tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl
Downloading https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl (151.3 MB)
Requirement already satisfied: numpy in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (1.19.2)
Requirement already satisfied: pillow in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (8.0.1)
Requirement already satisfied: bravado in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (11.0.2)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (2020.4)
Requirement already satisfied: aiohttp in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (3.7.3)
Requirement already satisfied: print-nanny-client in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (0.1.3)
Requirement already satisfied: websockets in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (8.1)
Requirement already satisfied: backoff==1.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (1.10.0)
Requirement already satisfied: aioprocessing==1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (1.1.0)
Requirement already satisfied: multiprocessing-logging==0.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (0.3.1)
Requirement already satisfied: typing_extensions in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.1.0) (3.7.4.3)
Requirement already satisfied: flatbuffers~=1.12.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.12)
Requirement already satisfied: termcolor~=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.1.0)
Requirement already satisfied: wheel~=0.35 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (0.36.2)
Requirement already satisfied: grpcio~=1.32.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.32.0)
Requirement already satisfied: h5py~=2.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (2.10.0)
Requirement already satisfied: tensorboard~=2.4 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (2.4.0)
Requirement already satisfied: six~=1.15.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.15.0)
Requirement already satisfied: tensorflow-estimator<2.5.0,>=2.4.0rc0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (2.4.0)
Requirement already satisfied: gast==0.3.3 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (0.3.3)
Requirement already satisfied: absl-py~=0.10 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (0.11.0)
Requirement already satisfied: google-pasta~=0.2 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (0.2.0)
Requirement already satisfied: opt-einsum~=3.3.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (3.3.0)
Requirement already satisfied: astunparse~=1.6.3 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.6.3)
Requirement already satisfied: protobuf>=3.9.2 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (3.14.0)
Requirement already satisfied: wrapt~=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.12.1)
Requirement already satisfied: keras-preprocessing~=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.1.2)
Requirement already satisfied: werkzeug>=0.11.15 in /home/pi/oprint/lib/python3.7/site-packages (from tensorboard~=2.4->tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.0.1)
Requirement already satisfied: requests<3,>=2.21.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorboard~=2.4->tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (2.25.1)
Requirement already satisfied: tensorboard-plugin-wit>=1.6.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorboard~=2.4->tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (1.7.0)
Requirement already satisfied: markdown>=2.6.8 in /home/pi/oprint/lib/python3.7/site-packages (from tensorboard~=2.4->tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (3.1.1)
Requirement already satisfied: setuptools>=41.0.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorboard~=2.4->tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (51.1.0.post20201221)
Requirement already satisfied: google-auth-oauthlib<0.5,>=0.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from tensorboard~=2.4->tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.1.0) (0.4.2)
[... cut ...]
Created wheel for OctoPrint-Nanny: filename=OctoPrint_Nanny-0.1.0-py3-none-any.whl size=13180098 sha256=5da94f322da3500dfeca6a2239304436ebf5f0bbffb2c97dc03ed852a3ed7f72
Stored in directory: /tmp/pip-ephem-wheel-cache-z3jzrnqd/wheels/71/b3/95/4be3317739eb167b5edc20d873860d9f459d78c7c3d6497557
Successfully built OctoPrint-Nanny
Installing collected packages: OctoPrint-Nanny
Attempting uninstall: OctoPrint-Nanny
Found existing installation: OctoPrint-Nanny 0.1.0
Uninstalling OctoPrint-Nanny-0.1.0:
Successfully uninstalled OctoPrint-Nanny-0.1.0
Successfully installed OctoPrint-Nanny-0.1.0
Fait !`

arut16, 2020-12-27:
Maybe there some missing libreries...

EuKeck, 2020-12-28:
Applies to me as well.
I am not able to install the plugin on freshly installed  OctoPi Version 0.18.0 (RC1)
(using a Raspberry Pi 2 Model B Rev 1.1)

[image]

Best regards 
E. Keck

DrDrache, 2020-12-28:
> Applies to me as well.
> I am not able to install the plugin on freshly installed OctoPi Version 0.18.0 (RC1)
> (using a Raspberry Pi 2 Model B Rev 1.1)
> 
> [image]
> 
> Best regards
> E. Keck

same here - same build (octopi .18 RC1)

leigh-johnson, 2020-12-28:
Thank you for the version/platform info! Reproduced this issue, fix incoming in 0.1.1. :bowing_woman:

leigh-johnson, 2021-01-08:
:wave: Hey y'all! The fix for this ended up in release v0.2.0. Please re-open if this does not work for you. Thanks!

https://github.com/bitsy-ai/octoprint-nanny-plugin/releases/tag/v0.2.0

closed 2021-01-08 by leigh-johnson