bitsy-ai/octoprint-nanny-plugin #19: Cannot install from GitHub sip link

sonicalchemy, 2020-12-27:
when I try to install from the Octoprint GUI using the github sip link I get this message;

ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-VMzb_n/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-VMzb_n/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-req-build-VMzb_n/pip-egg-info
cwd: /tmp/pip-req-build-VMzb_n/
Complete output (6 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-req-build-VMzb_n/setup.py", line 49
raise Exception(f'OctoPrint Nanny does not support {arch} architechture. Please open a Github issue.')
^
SyntaxError: invalid syntax
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

mobilegmYT, 2020-12-28:
Yep, same issue with me.

mobilegmYT, 2020-12-28:
Installing plugin from https://github.com/bitsy-ai/octoprint-nanny-plugin/archive/main.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpxKWmkc/octoprint-nanny-plugin-main.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpxKWmkc/octoprint-nanny-plugin-main.zip
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python2 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-vFsl_z/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-vFsl_z/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-req-build-vFsl_z/pip-egg-info
cwd: /tmp/pip-req-build-vFsl_z/
Complete output (6 lines):
Traceback (most recent call last):
File "<string>", line 1, in <module>
File "/tmp/pip-req-build-vFsl_z/setup.py", line 49
raise Exception(f'OctoPrint Nanny does not support {arch} architechture. Please open a Github issue.')
^
SyntaxError: invalid syntax
----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

leigh-johnson, 2020-12-28:
Hey y'all, this plugin only supports Python 3 and ARMv7 or AARCH64 cpu architectures (Raspberry Pi 4). Your OctoPrint installation is using Python2, and you're running on an older generation Raspberry Pi (ARMv6 or lower). 

I will add a note about these requirements to the FAQ and plugin info page, I apologize that you wasted some time on this! :bowing_woman: 

If you're looking to upgrade, I totally recommend a 4GB/8GB Raspberry Pi.

leigh-johnson, 2020-12-28:
Thanks again for writing in! Tagging this issue to train an auto-responder for this issue. I can tell it's going to be a common one. :joy:

closed 2020-12-28 by leigh-johnson

mobilegmYT, 2020-12-28:
Ah ok. I'm using a PI 3 which has an armv8 processor I believe, so I probably just need to update to python3. Thanks for the help!

leigh-johnson, 2020-12-28:
Gotcha! Hope all goes smoothly with the upgrade :crossed_fingers: 

If you have sec, dump your cpuinfo here for posterity.

```bash
$ cat /proc/cpuinfo
```

I haven't done any testing on Raspberry Pi 3 yet, so this is super helpful!

mobilegmYT, 2020-12-28:
Here you go!

`processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 2a22082
Serial          : 0000000083593018
Model           : Raspberry Pi 3 Model B Rev 1.2
`

mobilegmYT, 2020-12-28:
Ayyy it worked! For anyone looking to get the python3 version, see https://octoprint.org/blog/2020/09/10/upgrade-to-py3/.