bitsy-ai/octoprint-nanny-plugin #29: Still unable to install plugin

EuKeck, 2021-01-08:
Installation of Plugin in Version  0.2.0 still does not succeed.

_Originally posted by @EuKeck in https://github.com/bitsy-ai/octoprint-nanny-plugin/issues/18#issuecomment-751538030_

alxjrvs, 2021-01-10:
Also seeing this.

leigh-johnson, 2021-01-10:
Did you restart the OctoPrint server after installing the plugin?

dzervas, 2021-01-10:
*SOLVED*: On my raspi I needed to run `sudo apt-get install libatlas-base-dev`. For more info to solve this issue: https://numpy.org/devdocs/user/troubleshooting-importerror.html

---

I also have this bug.
- Press install from "Get more..."
- Wait
- Restart OctoPrint (I think I pressed the button - but maybe it auto restarted)

I'm on OctoPi 0.17.0 with OctoPrint 1.5.2 and Python 3.7.3, trying to install 0.2.0 Nanny

Logs:
```
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_nanny/__init__.py", line 7, in <module>
    from .plugins import OctoPrintNannyPlugin
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_nanny/plugins.py", line 25, in <module>
    import numpy as np
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/__init__.py", line 140, in <module>
    from . import core
  File "/home/pi/oprint/lib/python3.7/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.7 from "/home/pi/oprint/bin/python"
  * The NumPy version is: "1.19.5"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory
```

PR #35 (2021-01-10): Install platform dependencies 
Closes #29
PR #36 (2021-01-10): release v0.2.3
* Closes #29 #34 #33
closed 2021-01-10 by leigh-johnson