bitsy-ai/octoprint-nanny-plugin #34: Install process for RPi 4 2 GB Octopi

CTFishUSA, 2021-01-10:
Trying the install this evening on my RPI 4 2GB that was built with an Octopi image.

First error: 
AttributeError: 'tuple' object has no attribute 'machine'

This is surely a dependency problem.  I will give a shot at update / upgrade, and check again.  Then I'll start pulling in the dependent packages manually ...

CTFishUSA, 2021-01-10:
No dice.  Same error.  Completed the update and am now at 64-bit.

CTFishUSA, 2021-01-10:
Next try:
Unzip and run the install.sh from prompt

pi@octopi:/tmp $ ls -l
total 8
drwx------ 3 root root 4096 Jan  9 21:17 systemd-private-728fd60328a14664b8845f7072bec57a-systemd-timesyncd.service-xqT5Bx
drwx------ 2 pi   pi   4096 Jan  9 21:19 tmpCsIVbJ
pi@octopi:/tmp $ cd tmpCsIVbJ/
pi@octopi:/tmp/tmpCsIVbJ $ ls
octoprint-nanny-plugin-main.zip
pi@octopi:/tmp/tmpCsIVbJ $ unzip
UnZip 6.00 of 20 April 2009, by Debian. Original by Info-ZIP.

Usage: unzip [-Z] [-opts[modifiers]] file[.zip] [list] [-x xlist] [-d exdir]
  Default action is to extract files in list, except those in xlist, to exdir;
  file[.zip] may be a wildcard.  -Z => ZipInfo mode ("unzip -Z" for usage).

  -p  extract files to pipe, no messages     -l  list files (short format)
  -f  freshen existing files, create none    -t  test compressed archive data
  -u  update files, create if necessary      -z  display archive comment only
  -v  list verbosely/show version info       -T  timestamp archive to latest
  -x  exclude files that follow (in xlist)   -d  extract files into exdir
modifiers:
  -n  never overwrite existing files         -q  quiet mode (-qq => quieter)
  -o  overwrite files WITHOUT prompting      -a  auto-convert any text files
  -j  junk paths (do not make directories)   -aa treat ALL files as text
  -U  use escapes for all non-ASCII Unicode  -UU ignore any Unicode fields
  -C  match filenames case-insensitively     -L  make (some) names lowercase
  -X  restore UID/GID info                   -V  retain VMS version numbers
  -K  keep setuid/setgid/tacky permissions   -M  pipe through "more" pager
See "unzip -hh" or unzip.txt for more help.  Examples:
  unzip data1 -x joe   => extract all files except joe from zipfile data1.zip
  unzip -p foo | more  => send contents of foo.zip via pipe into program more
  unzip -fo foo ReadMe => quietly replace existing ReadMe if archive file newer
pi@octopi:/tmp/tmpCsIVbJ $ unzip octoprint-nanny-plugin-main.zip 
Archive:  octoprint-nanny-plugin-main.zip
84e9837c70d19a29f58b6b5958c634be62890d6b
   creating: octoprint-nanny-plugin-main/
  inflating: octoprint-nanny-plugin-main/.editorconfig  
  inflating: octoprint-nanny-plugin-main/.gitignore  
  inflating: octoprint-nanny-plugin-main/LICENSE  
  inflating: octoprint-nanny-plugin-main/MANIFEST.in  
  inflating: octoprint-nanny-plugin-main/Makefile  
  inflating: octoprint-nanny-plugin-main/README.md  
  inflating: octoprint-nanny-plugin-main/babel.cfg  
  inflating: octoprint-nanny-plugin-main/dev-requirements.txt  
   creating: octoprint-nanny-plugin-main/docs/
   creating: octoprint-nanny-plugin-main/docs/images/
  inflating: octoprint-nanny-plugin-main/docs/images/logo.jpg  
  inflating: octoprint-nanny-plugin-main/docs/images/screenshot_1.png  
  inflating: octoprint-nanny-plugin-main/docs/images/screenshot_2.png  
  inflating: octoprint-nanny-plugin-main/docs/images/screenshot_3.png  
  inflating: octoprint-nanny-plugin-main/docs/images/screenshot_4.png  
  inflating: octoprint-nanny-plugin-main/docs/images/screenshot_5.png  
   creating: octoprint-nanny-plugin-main/extras/
  inflating: octoprint-nanny-plugin-main/extras/README.txt  
  inflating: octoprint-nanny-plugin-main/extras/octoprint_nanny.md  
   creating: octoprint-nanny-plugin-main/hooks/
  inflating: octoprint-nanny-plugin-main/hooks/pre-commit  
  inflating: octoprint-nanny-plugin-main/install.sh  
  inflating: octoprint-nanny-plugin-main/nginx.conf  
  inflating: octoprint-nanny-plugin-main/octoprint-nanny.workspace.code-workspace  
   creating: octoprint-nanny-plugin-main/octoprint_nanny/
  inflating: octoprint-nanny-plugin-main/octoprint_nanny/__init__.py  
   creating: octoprint-nanny-plugin-main/octoprint_nanny/clients/
 extracting: octoprint-nanny-plugin-main/octoprint_nanny/clients/__init__.py  
  inflating: octoprint-nanny-plugin-main/octoprint_nanny/clients/honeycomb.py  
  inflating: octoprint-nanny-plugin-main/octoprint_nanny/clients/mqtt.py  
  inflating: octoprint-nanny-plugin-main/octoprint_nanny/clients/rest.py  
  inflating: octoprint-nanny-plugin-main/octoprint_nanny/clients/websocket.py  
[... 109 lines cut ...]
pi@octopi:/tmp/tmpCsIVbJ/octoprint-nanny-plugin-main $ chmod +x install.sh
pi@octopi:/tmp/tmpCsIVbJ/octoprint-nanny-plugin-main $ ./install.sh
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
[sudo] password for pi: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libatlas-base-dev is already the newest version (3.10.3-8+rpi1).
The following packages were automatically installed and are no longer required:
  libllvm8 rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cmake is already the newest version (3.13.4-1).
imagemagick is already the newest version (8:6.9.10.23+dfsg-2.1+deb10u1).
libjpeg62-turbo-dev is already the newest version (1:1.5.2-2+deb10u1).
libv4l-dev is already the newest version (1.16.3-3).
subversion is already the newest version (1.10.4-1+deb10u1).
ffmpeg is already the newest version (7:4.1.6-1~deb10u1+rpt1).
The following packages were automatically installed and are no longer required:
  libllvm8 rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

CTFishUSA, 2021-01-10:
Next, i'm getting pip ... surely I had this already?

dzervas, 2021-01-10:
Are you running python3 octoprint?

CTFishUSA, 2021-01-10:
Worked through the dev_requirements.  Now to try the plugin.

CTFishUSA, 2021-01-10:
> Are you running python3 octoprint?

How?

CTFishUSA, 2021-01-10:
Found this ... https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

CTFishUSA, 2021-01-10:
OK, I ran the upgrade-to-py3 script from above.  Octoprint is now at Python3. 
Also updated Octolapse once the server came back.

Plugin installed using the plugin manager, but errors out with
 NameError: name 'tensorflow' is not defined
I think that's someone else's bug

CTFishUSA, 2021-01-10:
I'll check when 0.2.3 is out, looks like the tensorflow is being addressed in that release.

leigh-johnson, 2021-01-10:
Thanks for the report! Let me know how the 0.2.3 installation goes.

closed 2021-01-10 by leigh-johnson