bitsy-ai/octoprint-nanny-plugin #45: Fail with a clean error on Python 2

leigh-johnson, 2021-01-10:
Many folks are trying to install Print Nanny in a Python 2 environment. This fails with a non-obvious error like the one below. 

In the installer, do a check for Python version and explicitly fail with a link to the FAQ if Python 2 is detected. https://help.print-nanny.com/faq/do-you-support-python-2/

Test a pre-release of this logic; some folks are reporting their OctoPrint setup still uses Python2's pip module after migrating to Python3. 

https://discordapp.com/channels/773452324692688956/783897526125199380/797880941522452520

[image]

seacred, 2021-01-10:
[image]
i'm unable to manually install the plugin via pop3.
dont know if it matters but wouldn't be pluginmanager unable to install that too, by using pip3?

leigh-johnson, 2021-01-10:
Hi @seacred, have you tired following this guide to upgrade your OctoPrint installation to Python 3?
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

OctoPrint uses a virtual environment, so you'll need to upgrade its virtual environment. `pip3 install` (with no virtual environment active) will try to install using your system interpreter and pip3 module, which won't work. Let me know if that does not make sense to you.

seacred, 2021-01-10:
Did the Upgrade to python 3 already, like i mentioned in your bugbounty discord -> @Strohm.
Tried some different things to make octoprint use pip3 but with no success at all.

leigh-johnson, 2021-01-10:
> Did the Upgrade to python 3 already, like i mentioned in your bugbounty discord -> @strohm.
> Tried some different things to make octoprint use pip3 but with no success at all.

Aw, sorry - didn't realize you were part of the Discord crew. Thank you so much for trying this out and updating me on progress (or lack thereof :joy:)

I'll dig deeper into reproducing this issue and see if I can figure out a fix, but it might take a few weeks for me to prioritize. I'll have to setup a new Pi with OctoPi 0.17.0 and follow the upgrade path myself to grok what's going on. 

In the meantime, if you're itching to try the plugin **now**, OctoPi 0.18.0 defaults to Python3 support. You can gzip /home/pi/.octoprint to backup your current OctoPrint settings and data.

closed 2021-01-21 by leigh-johnson