bitsy-ai/octoprint-nanny-plugin #54: Still unable to install plugin

EuKeck, 2021-01-10:
... what about making an integration test, that tries to install the plugin on a fresh octoprint installation ???
Currently I don´t know if the access problems (files accessed by package installer) results from the pervious failed installations or are a systematic error in your installer script.

Error from plugin version 0.2.6:

Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog==0.10.4->OctoPrint->OctoPrint-Nanny==0.2.6) (0.1.2)
Requirement already satisfied: sgmllib3k in /home/pi/oprint/lib/python3.7/site-packages (from feedparser<7,>=6.0.2->OctoPrint->OctoPrint-Nanny==0.2.6) (1.0.0)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->OctoPrint-Nanny==0.2.6) (2.0)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoPrint-Nanny==0.2.6) (2.8.0)
Requirement already satisfied: ifaddr in /home/pi/oprint/lib/python3.7/site-packages (from zeroconf<0.25,>=0.24->OctoPrint->OctoPrint-Nanny==0.2.6) (0.1.7)
Building wheels for collected packages: OctoPrint-Nanny
Building wheel for OctoPrint-Nanny (setup.py): started
Building wheel for OctoPrint-Nanny (setup.py): finished with status 'error'
Running setup.py clean for OctoPrint-Nanny
ERROR: Command errored out with exit status 1:
command: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-x1dxlpkk/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-x1dxlpkk/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-ysjqb2xg
cwd: /tmp/pip-req-build-x1dxlpkk/
Complete output (113 lines):
Found packages: {'octoprint_nanny', 'octoprint_nanny.clients', 'octoprint_nanny.utils'}
running bdist_wheel
running build
running build_py
creating build
creating build/lib
creating build/lib/octoprint_nanny
copying octoprint_nanny/errors.py -> build/lib/octoprint_nanny
copying octoprint_nanny/__init__.py -> build/lib/octoprint_nanny
copying octoprint_nanny/predictor.py -> build/lib/octoprint_nanny
copying octoprint_nanny/manager.py -> build/lib/octoprint_nanny
copying octoprint_nanny/plugins.py -> build/lib/octoprint_nanny
creating build/lib/octoprint_nanny/clients
copying octoprint_nanny/clients/__init__.py -> build/lib/octoprint_nanny/clients
copying octoprint_nanny/clients/rest.py -> build/lib/octoprint_nanny/clients
copying octoprint_nanny/clients/mqtt.py -> build/lib/octoprint_nanny/clients
copying octoprint_nanny/clients/websocket.py -> build/lib/octoprint_nanny/clients
copying octoprint_nanny/clients/honeycomb.py -> build/lib/octoprint_nanny/clients
creating build/lib/octoprint_nanny/utils
copying octoprint_nanny/utils/__init__.py -> build/lib/octoprint_nanny/utils
copying octoprint_nanny/utils/encoder.py -> build/lib/octoprint_nanny/utils
copying octoprint_nanny/utils/visualization.py -> build/lib/octoprint_nanny/utils
running egg_info
creating OctoPrint_Nanny.egg-info
writing OctoPrint_Nanny.egg-info/PKG-INFO
writing dependency_links to OctoPrint_Nanny.egg-info/dependency_links.txt
writing entry points to OctoPrint_Nanny.egg-info/entry_points.txt
writing requirements to OctoPrint_Nanny.egg-info/requires.txt
writing top-level names to OctoPrint_Nanny.egg-info/top_level.txt
writing manifest file 'OctoPrint_Nanny.egg-info/SOURCES.txt'
reading manifest file 'OctoPrint_Nanny.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*' under directory 'octoprint_nanny/translations'
writing manifest file 'OctoPrint_Nanny.egg-info/SOURCES.txt'
creating build/lib/octoprint_nanny/data
creating build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/0.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/0.pre.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/1.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/1.pre.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/2.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/2.pre.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/3.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/3.pre.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/4.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/4.pre.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/5.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/5.pre.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/6.post.jpg -> build/lib/octoprint_nanny/data/images
copying octoprint_nanny/data/images/6.pre.jpg -> build/lib/octoprint_nanny/data/images
creating build/lib/octoprint_nanny/data/tflite-print3d-2020-10-23T18:00:41.136Z
copying octoprint_nanny/data/tflite-print3d-2020-10-23T18:00:41.136Z/dict.txt -> build/lib/octoprint_nanny/data/tflite-print3d-2020-10-23T18:00:41.136Z
[... 179 lines cut ...]
self.distribution.run_command(command)
File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
cmd_obj.run()
File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
self.run_command(cmd_name)
File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
self.distribution.run_command(command)
File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
cmd_obj.run()
File "/tmp/pip-req-build-x1dxlpkk/setup.py", line 153, in run
self.run_command(command)
File "/tmp/pip-req-build-x1dxlpkk/setup.py", line 149, in run_command
return self.run_command(command, sudo=True)
File "/tmp/pip-req-build-x1dxlpkk/setup.py", line 144, in run_command
command, p.returncode
RuntimeError: PLATFORM_INSTALL Command ['sudo', 'apt-get', 'update'] failed: exit code:1
----------------------------------------
ERROR: Command errored out with exit status 1: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-x1dxlpkk/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-x1dxlpkk/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-ci4p1c09/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.7/OctoPrint-Nanny Check the logs for full command output.
Fehler!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

(Note: A reboot of the raspberry pi and new installation of the plugin had no effect.) 

Regards,
E.Keck

thomasctobin, 2021-01-12:
I have the same issue, https://cdn.discordapp.com/attachments/783897526125199380/798343260670132264/message.txt
Posted log on the discord group.

CTFishUSA, 2021-01-12:
What platform are you using?  If Pi, are you on the 64-bit or 32-bit kernel?  https://medium.com/for-linux-users/how-to-make-your-raspberry-pi-4-faster-with-a-64-bit-kernel-77028c47d653

thomasctobin, 2021-01-12:
32bit, with PY3.

CTFishUSA, 2021-01-12:
I've now learned the difference between user land and kernel ... Leigh would be pleased.  I will pursue octopi 64 bit (64/64) once it's stable, but stick with 64kernel and 32 user for now.  I did successfully install so not sure why you're getting the error.  You might try installing dependencies outside of octoprint first to look for things in the log/prompt stream.

EuKeck, 2021-01-12:
Raspberry Pi 2 Model B Rev 1.1
--> 32 bit
OctoPi Version 0.18.0 (RC1)

CTFishUSA, 2021-01-12:
> Raspberry Pi 2 Model B Rev 1.1
> --> 32 bit
> OctoPi Version 0.18.0 (RC1)

Do you think that will run tensor flow?

thomasctobin, 2021-01-12:
I'm on a rasp pi 4b, should be able to run it :/, a pi 2 though, i'm doubtful.

EuKeck, 2021-01-12:
> Do you think that will run tensor flow?

May be.  (This pi model has 4 Cores and 1GB RAM) I don´t use WLAN just pain LAN connection. What are the recommendations?  I just wanted to try it out.

Regards, E. Keck

matthieudolci, 2021-01-19:
This will allow to install the plugin. It is not a secure thing to do tho.

`visudo`
 add line 
`pi ALL=(ALL:ALL) NOPASSWD:ALL`
`ctrl+x`

leigh-johnson, 2021-01-20:
> I've now learned the difference between user land and kernel ... Leigh would be pleased. I will pursue octopi 64 bit (64/64) once it's stable, but stick with 64kernel and 32 user for now. I did [...]

This made me so happy! Print Nanny is certainly a learning experience / adventure for me, and it brings me joy to see other folks learning cool new things too. :heart: 

I'm going to dig into this installer issue today. 

The root issue is that Print Nanny requires some platform dependencies prior to installation. I tried to be clever and install these in a sub-process during `pip install`, but not everyone's system is configured to permit passwordless or promptless sudo. This breaks the UI installation, and probably just seems weird at first glance ("why is this pip module prompting me for my sudo password???").

leigh-johnson, 2021-01-21:
:bowing_woman: I rewrote libatlas installation to install platform dependencies with Ansible (without sudo). Please re-open this issue if version >=0.3.3 still doesn't work for you. Thanks!

PR #72 (2021-01-20): Sudo-less libatlas install
* closes #54
closed 2021-01-21 by leigh-johnson