bitsy-ai/octoprint-nanny-plugin #55: Unable to install plugin via manager - Octopi .18 RC2 32bit

DrDrache, 2021-01-13:
I switched back to 32bit Octopi .18 as the new 64bit is seriously broken - will go back to 64bit testing when a 64bit version is stable again.

I attempted to install the plugin and get this message (full console logs attached):

2021-01-13 13:39:11,155 > Building wheel for OctoPrint-Nanny (setup.py): started
2021-01-13 13:39:12,465 > Building wheel for OctoPrint-Nanny (setup.py): finished with status 'error'
2021-01-13 13:39:12,481 > Running setup.py clean for OctoPrint-Nanny
2021-01-13 13:39:12,985 ! ERROR: Command errored out with exit status 1:
2021-01-13 13:39:12,986 ! command: /home/pi/oprint/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-18oufm_y/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-18oufm_y/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-pjn3ekc4
2021-01-13 13:39:12,986 ! cwd: /tmp/pip-req-build-18oufm_y/

2021-01-13 13:39:12,996 ! Running PLATFORM_INSTALL command: ['apt-get', 'update']
2021-01-13 13:39:12,996 ! PLATFORM_INSTALL Command output: b'Reading package lists...\nE: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)\nE: Unable to lock directory /var/lib/apt/lists/\nW: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)\nW: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)\n'
2021-01-13 13:39:12,996 ! Running PLATFORM_INSTALL command: ['sudo', 'apt-get', 'update']
2021-01-13 13:39:12,996 ! PLATFORM_INSTALL Command output: b'sudo: no tty present and no askpass program specified\n'

[plugin_pluginmanager_console.log](https://github.com/bitsy-ai/octoprint-nanny-plugin/files/5808956/plugin_pluginmanager_console.log)

nmsmith22389, 2021-01-15:
I'm having the same problem.

MambaRoja, 2021-01-23:
unable to install, no error message:

Installing plugin "OctoPrint Nanny" from https://github.com/bitsy-ai/octoprint-nanny-plugin/archive/main.zip.../home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpgn2m80ob/octoprint-nanny-plugin-main.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simpleProcessing /tmp/tmpgn2m80ob/octoprint-nanny-plugin-main.zipRequirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (1.5.3)Collecting tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whlDownloading https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl (151.3 MB)Requirement already satisfied: numpy in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (1.19.4)Requirement already satisfied: pillow in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (8.0.1)Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (2020.4)Requirement already satisfied: aiohttp in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (3.7.3)Requirement already satisfied: print-nanny-client~=0.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (0.3.1)Requirement already satisfied: websockets in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (8.1)Requirement already satisfied: backoff==1.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (1.10.0)Requirement already satisfied: aioprocessing==1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (1.1.0)Requirement already satisfied: multiprocessing-logging==0.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (0.3.1)Requirement already satisfied: jwt in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (1.1.0)Requirement already satisfied: paho-mqtt==1.5.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (1.5.1)Requirement already satisfied: honeycomb-beeline in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (2.16.1)Requirement already satisfied: typing_extensions in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Nanny==0.3.4) (3.7.4.3)Requirement already satisfied: grpcio~=1.32.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (1.32.0)Requirement already satisfied: protobuf>=3.9.2 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (3.14.0)Requirement already satisfied: h5py~=2.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (2.10.0)Requirement already satisfied: flatbuffers~=1.12.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (1.12)Requirement already satisfied: astunparse~=1.6.3 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (1.6.3)Requirement already satisfied: termcolor~=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (1.1.0)Requirement already satisfied: keras-preprocessing~=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (1.1.2)Requirement already satisfied: six~=1.15.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (1.15.0)Requirement already satisfied: opt-einsum~=3.3.0 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (3.3.0)Requirement already satisfied: google-pasta~=0.2 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (0.2.0)Requirement already satisfied: wheel~=0.35 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ https://github.com/bitsy-ai/tensorflow-arm-bin/releases/download/v2.4.0/tensorflow-2.4.0-cp37-none-linux_armv7l.whl->OctoPrint-Nanny==0.3.4) (0.36.2)Requirement already satisfied: wrapt~=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from tensorflow@ http [... cut ...] 13267797 sha256=f5702acbec650fa12ef67ba5c379be25cadb1bfa1afeda469f8cbd0defd79594Stored in directory: /tmp/pip-ephem-wheel-cache-f5vyyt84/wheels/73/8d/55/3fe495164b18e93c657c99db2927b43657359adce3bcfee315Successfully built OctoPrint-NannyInstalling collected packages: OctoPrint-NannyAttempting uninstall: OctoPrint-NannyFound existing installation: OctoPrint-Nanny 0.3.4Uninstalling OctoPrint-Nanny-0.3.4:Successfully uninstalled OctoPrint-Nanny-0.3.4Successfully installed OctoPrint-Nanny-0.3.4Done!

MambaRoja, 2021-01-23:
[octoprint.log](https://github.com/bitsy-ai/octoprint-nanny-plugin/files/5860995/octoprint.log)

leigh-johnson, 2021-01-25:
Hey @MambaRoja, thanks for the logs. Can you SSH into your Raspberry Pi and run:

`sudo apt-get update && sudo apt-get install libatlas3-base`

This will install ATLAS (Automatically Tuned Linear Alebra Software) C libraries. 

After that, reinstall the Print Nanny plugin and restart OctoPrint.

leigh-johnson, 2021-01-29:
Heyo, closing this for now. Please re-open if you continue to experience installation issues! I've also added these steps to the OctoPrint Setup guide: https://help.print-nanny.com/octoprint-plugin-setup/#install-prerequisites

Thank you for the report! bowing_woman:

closed 2021-01-29 by leigh-johnson

MambaRoja, 2021-01-30:
Sorry for the delay, it is working now!

Thanks

MambaRoja