bitsy-ai/octoprint-nanny-plugin #59: Exceptions thrown in manager _handle_print_start

leigh-johnson, 2021-01-18:
```
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/leigh/projects/octoprint-nanny-plugin/octoprint_nanny/manager.py", line 263, in _telemetry_worker
    asyncio.ensure_future(self._telemetry_queue_send_loop())
  File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
    return future.result()
  File "/home/leigh/projects/octoprint-nanny-plugin/octoprint_nanny/manager.py", line 436, in _telemetry_queue_send_loop
    await handler_fn(**event)
TypeError: _handle_print_start() got an unexpected keyword argument 'user_id'
```

thomasctobin, 2021-01-18:
Is this solving the install bugs we are having?

leigh-johnson, 2021-01-18:
Hey @thomasctobin, no - this fixes a bug experienced when starting a print (after installation succeeded). 

There are a couple of installation bugs out there. Can you double-check to see if you're impacted by one of these existing issues?

* `sudo apt-get` fails during install (permission error) 
https://github.com/bitsy-ai/octoprint-nanny-plugin/issues/53
https://github.com/bitsy-ai/octoprint-nanny-plugin/issues/55
https://github.com/bitsy-ai/octoprint-nanny-plugin/issues/54
* 64-bit installation fails (need more info)
https://github.com/bitsy-ai/octoprint-nanny-plugin/issues/23

commit 5974169d (2021-01-19): Print progress (#60)

* fixes #59

* fix exception thrown in print progress publisher

* fix printer profile REST payload

* add new fields to printer profile request, fix metadata missing

* s/dt/created_dt

* serialize octoprint_device id as a string in multipart form upload; see https://github.com/aio-libs/aiohttp/issues/3652

* mute Zchange warnings, fix print progress event format

* lint

* first pass on capture snapshot command

* send snapshot with remote command status update

* add lambda handler

* send device metadata with remote command status update

* lint

* add extra exception [... cut]
PR #61 (2021-01-19): Release v0.3.0
### Features

* Receive notifications on percentage progress/completion of a print
* Added UI notification system to Print Nanny Webapp (new Vue app)
* Added a TypeScript API client (unpublished)
* Added notification settings and history to Print Nanny Webapp

A more detailed write up is coming in the first ever Print Nanny newsletter!

### Bug Fixes
* fixes #59
* Fixes a bunch exceptions appearing in HoneyComb

[image]
commit 45fcc11f (2021-01-19): Release v0.3.0 (#61)

* Print progress (#60)

* fixes #59

* fix exception thrown in print progress publisher

* fix printer profile REST payload

* add new fields to printer profile request, fix metadata missing

* s/dt/created_dt

* serialize octoprint_device id as a string in multipart form upload; see https://github.com/aio-libs/aiohttp/issues/3652

* mute Zchange warnings, fix print progress event format

* lint

* first pass on capture snapshot command

* send snapshot with remote command status update

* add lambda handler

* send device metadata with remote command status update

* lin [... cut]
closed 2021-01-19 by leigh-johnson