bitsy-ai/octoprint-nanny-plugin #63: setup.py os.uname().machine is not a good fit for Pi4 with 64bit kernel and 32bit userland -- tensorflow install fail at 0.3.0

CTFishUSA, 2021-01-20:
setup.py

arch = os.uname().machine
is returning the kernel version of my Pi4 (aarch64) but  I have the 32bit userland.  

This causes an install failure of tensorflow.  Workaround is to hardcode arch = "armv7l" for this situation (64bit kernel, 32bit userland).  

If practical, find alternative test for arch

CTFishUSA, 2021-01-20:
Here's my suggestion.  Right after:

arch = os.uname().machine

Add this statement (and some situational comments):

if platform.architecture()[0]=='32bit': arch="armv7l"

leigh-johnson, 2021-01-20:
Thank you so much for following up on this! :bowing_woman: Especially appreciate that you took the time to debug and figure out an alternative. 

Hopefully this covers most cases. :crossed_fingers: 
https://github.com/bitsy-ai/octoprint-nanny-plugin/commit/acf3d9cc1accb9a1050564a2e6cc87c799b43c0d#diff-60f61ab7a8d1910d86d9fda2261620314edcae5894d5aaa236b821c7256badd7R61

commit 362b1a5f (2021-01-20): closes #63; handle installation on systems with a 64-bit CPU and kernel space, but 32-bit userland (#64)
PR #65 (2021-01-20): Release v0.3.1
* Closes #63 
* Gracefully re-initialize worker threads when certain OctoPrint settings stored in thread-local state are changed (device registration and auth token).
commit 727bbb11 (2021-01-20): Release v0.3.1 (#65)

* closes #63; handle installation on systems with a 64-bit CPU and kernel space, but 32-bit userland (#64)

* Re init worker threads (#66)

* improve mqtt client logging and add backoff

* halt and re-initialize worker threads on new auth and device registration settings

* fix dupe honeycomb dataset

* fix misc exceptions

* fixup halt event handling

* nit

* Bump version: 0.3.0 → 0.3.1
closed 2021-01-20 by leigh-johnson