bitsy-ai/octoprint-nanny-plugin #86: MQTT client disconnects with rc=4 rc=5

leigh-johnson, 2021-01-24:
- OctoPrint version: 1.5.x
- Print Nanny version: 0.3.4
- OctoPi (or other OS) version: 0.18-rc2

**Describe the bug**
Authorization token (JWT) passed during connect is rejected by the MQTT gateway. 
```
2021-01-21 22:26:47,482 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - MQTTClient._on_connect called with client=<paho.mqtt.client.Client object at 0xdee3bef0> userdata=None rc=4
2021-01-21 22:26:47,482 - octoprint.plugins.octoprint_nanny.clients.mqtt - ERROR - Connection refused by MQTT broker with reason code rc=4
2021-01-21 22:26:47,489 - octoprint.plugins.octoprint_nanny.clients.mqtt - WARNING - Device disconnected from MQTT bridge client=<paho.mqtt.client.Client object at 0xdee3bef0> userdata=None rc=5
2021-01-21 22:27:51,553 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - Device attempting to re-authenticate with MQTT broker (JWT probably expired)
2021-01-21 22:27:51,617 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - Creating JWT using RS256 from private key file /home/pi/.octoprint/data/octoprint_nanny/private_key.pem
2021-01-21 22:28:08,132 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - MQTTClient._on_connect called with client=<paho.mqtt.client.Client object at 0xdee3bef0> userdata=None rc=4
2021-01-21 22:28:08,132 - octoprint.plugins.octoprint_nanny.clients.mqtt - ERROR - Connection refused by MQTT broker with reason code rc=4
2021-01-21 22:28:08,138 - octoprint.plugins.octoprint_nanny.clients.mqtt - WARNING - Device disconnected from MQTT bridge client=<paho.mqtt.client.Client object at 0xdee3bef0> userdata=None rc=5
2021-01-21 22:30:16,243 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - Device attempting to re-authenticate with MQTT broker (JWT probably expired)
2021-01-21 22:30:16,358 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - Creating JWT using RS256 from private key file /home/pi/.octoprint/data/octoprint_nanny/private_key.pem
2021-01-21 22:30:48,878 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - MQTTClient._on_connect called with client=<paho.mqtt.client.Client object at 0xdee3bef0> userdata=None rc=4
2021-01-21 22:30:48,878 - octoprint.plugins.octoprint_nanny.clients.mqtt - ERROR - Connection refused by MQTT broker with reason code rc=4
2021-01-21 22:30:48,885 - octoprint.plugins.octoprint_nanny.clients.mqtt - WARNING - Device disconnected from MQTT bridge client=<paho.mqtt.client.Client object at 0xdee3bef0> userdata=None rc=5
```

Re-provisioning the device (generates a new key pair) yields the same result.

So far, only CTFish is the only person reporting this issue (please comment if you see the error above). I see some devices registered a heartbeat, so I think the culprit must be specific to CTFish's system configuration OR an unusual state he ended up in via enthusiastic testing.

**To Reproduce**
Can't reproduce this yet

**Expected behavior**
MQTT client should initialize and maintain a connection until JWT expires (60 minutes)

**Screenshots**
If applicable, add screenshots to help explain your problem.

leigh-johnson, 2021-01-24:
Investigation: client / server SSL version mismatch?

Steps to provide necessary info:

1. Run OctoPrint's python interpreter
```
pi@octopi:~/OctoPrint$ /home/pi/oprint/bin/python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```
2. Import `sys` and `ssl` modules
```
>>> import sys
>>> import ssl
```
3. Print sys.version, sys-platform just for reference
```
>>> "Python {:s} on {:s}".format(sys.version, sys.platform)
'Python 3.7.3 (default, Jul 25 2020, 13:03:44) \n[GCC 8.3.0] on linux'
```
4. Print default SSL options

```
>>> ctx0 = ssl.create_default_context()
>>> ctx0.options
<Options.OP_NO_SSLv3|OP_CIPHER_SERVER_PREFERENCE|OP_ENABLE_MIDDLEBOX_COMPAT|OP_NO_COMPRESSION|OP_ALL: -2108555180>
```

CTFishUSA, 2021-01-24:
pi@octopi:~/.octoprint/logs $ /home/pi/oprint/bin/python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import ssl
>>> "Python {:s} on {:s}".format(sys.version, sys.platform)
'Python 3.7.3 (default, Jul 25 2020, 13:03:44) \n[GCC 8.3.0] on linux'
>>> ctx0 = ssl.create_default_context()
>>> ctx0.options
<Options.OP_NO_SSLv3|OP_CIPHER_SERVER_PREFERENCE|OP_ENABLE_MIDDLEBOX_COMPAT|OP_NO_COMPRESSION|OP_ALL: -2108555180>
>>>

CTFishUSA, 2021-01-31:
Started working 1/30/2021 with 0.4.1.  Am not sure exactly why.

dzervas, 2021-02-04:
Unfortunately I'm on the same boat.

```bash
(oprint) pi@octopi:~/.octoprint $  /home/pi/oprint/bin/python3
Python 3.7.3 (default, Jul 25 2020, 13:03:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import ssl
>>> "Python {:s} on {:s}".format(sys.version, sys.platform)
'Python 3.7.3 (default, Jul 25 2020, 13:03:44) \n[GCC 8.3.0] on linux'
>>> ctx0 = ssl.create_default_context()
>>> ctx0.options
<Options.OP_NO_SSLv3|OP_CIPHER_SERVER_PREFERENCE|OP_ENABLE_MIDDLEBOX_COMPAT|OP_NO_COMPRESSION|OP_ALL: -2108555180>
>>>
```
I've attached the most part of my octoprint.log

[octoprint.log](https://github.com/bitsy-ai/octoprint-nanny-plugin/files/5922685/octoprint.log)

CTFishUSA, 2021-02-04:
Make sure you're using the latest version.  Leigh has added lots of additional logging that is captured via honeycomb/beeline.  I'm still broken but am at least providing failure data in the background.

dzervas, 2021-02-04:
By the way I solved this by tinkering with some environmental variables - I wanted to use the development environment. For more info https://github.com/bitsy-ai/octoprint-nanny-plugin/blob/main/Makefile#L57

leigh-johnson, 2021-02-04:
Please DO NOT tinker with the environment variables! These are only intended for developer use. Thanks!

@dzervas the instructions I provided you are for a developer setup.

dzervas, 2021-02-05:
@leigh-johnson forgot to ask the main question to @CTFishUSA: You're not trying to setup your own nanny instance, are you? custom urls n stuff (that's why I talked about env vars)

CTFishUSA, 2021-02-05:
No, working with Leigh's cloud as normal.  Chasing theory that this is metadata-garbling or time-induced refusal by gcp to authorize.  Have shared notes with Leigh and she's taking the lead.

CTFishUSA, 2021-02-06:
0.4.3, error persists

CTFishUSA, 2021-11-13:
Working fine now on latest build.  Not sure when it started, haven't been printing for a while.  Now back at it with latest releases across the board and it's working fine.

leigh-johnson, 2021-11-15:
:raised_hands: Thanks for reporting back in @CTFishUSA! I'm going to close this issue out for now, but let me know if you run into any problems! Welcome back :D

closed 2021-11-15 by leigh-johnson