bitsy-ai/octoprint-nanny-plugin #88: Unable to start the server after installation

Raptoaaah, 2021-01-24:
- OctoPrint version: 1.5.3 (Python 3.8)
- Print Nanny version: 0.3.4
- OctoPi (or other OS) version: Ubuntu 20.10

**Describe the bug**
Installation seems to work, but the server restarts afterwards and refuses to come up.

**To Reproduce**
Steps to reproduce the behavior:
1. Install plugin
2. See error

**Expected behavior**
OctoPrint should come up afterwards.

**Screenshots**
[image]

leigh-johnson, 2021-01-25:
Hey @Raptoaaah, can you attach the logs in ~/.octoprint/logs?

What is the output of...

```
$ ~/OctoPrint/venv/bin/python3 -c "import platform; print(platform.architecture())"
$  ~/OctoPrint/venv/bin/python3 -c "import os; print(os.uname().machine)"
$ ~/OctoPrint/venv/bin/python3 -c  "import tensorflow; print(tensorflow.__version__)"
$ ~/OctoPrint/venv/bin/python3 -c  "import numpy; print(numpy.__version__)"

```

Just as a head's up, you're the first person (that I know of) to try this on Ubuntu 20.10. It might take a bit of back/forth to debug the difference in system configuration.

leigh-johnson, 2021-01-25:
Just thought of this - you're probably missing ATLAS (Automatically Tuned Linear Alebra Software) C libraries.

Can you rSSH into your Raspberry Pi and run the following?
`sudo apt-get update && sudo apt-get install libatlas3-base` 

After that, re-install Print Nanny and restart OctoPrint. Let me know how that goes. 

Even if that works, I'd appreciate the diagnostic info mentioned above so I can setup a test environment. Thanks!

Raptoaaah, 2021-01-25:
Hey @leigh-johnson, thank you for your help. 

I ran your commands and this is what it tells about my environment:
('64bit', 'ELF')
x86_64
Does not work - I guess there is a problem with tensorflow (even only importing tensorflow fails)
1.19.5

My host is a Lenovo X201 i7-620m with 4G ram and 512GB SSD. 

ATLAS-lib was missing, thats true. As tensorflow seems to be broken I have to go deeper into this - reinstall and remove all the caches changes nothing. Actually I cannot be sure if this one is only present on my system alone or only a few, maybe a specific python version. I will go deeper into this, hope I can find the reason.
My exact Python version is 3.8.6, from the logs tensorflow should be 2.4.1. Without Tensorflow fixed I think there is no chance for the server to start and no direct plugin-related problem. I know this might be hard to reproduce on other systems, I try to provide a most complete outline of what I did to resolve it if I get it up and running again.

Sorry for the screenshot, I attached the logs here. If you need something else I try to get it for you.
[octoprint.log](https://github.com/bitsy-ai/octoprint-nanny-plugin/files/5865669/octoprint.log)

I'm kinda used to having problems regarding my more exotic envs, I'll try to help as much as possible.

Raptoaaah, 2021-01-25:
It seems to be likely that my system lacks AVX-support and therefore cannot run current TensorFlow versions. I'll try to change to a platform supporting current TensorFlow. Minimum requirement seems to be "second gen" Intel Core processor if using x86/AMD64 based system. Also "Celeron" and "Pentium" seem to lack AVX support.
For reference: https://en.wikipedia.org/wiki/Advanced_Vector_Extensions#CPUs_with_AVX

leigh-johnson, 2021-01-29:
Thank you for digging into this issue and sharing your findings! @Raptoaaah :bowing_woman: 

For now, I've added the `CPU Arch Not Supported` label. It's possible to compile TensorFlow without AVX instructions, but I'd need to research how to do this (and how to sanity-check the results). 

I build the ARM wheels for Raspberry Pi myself: https://github.com/bitsy-ai/tensorflow-arm-bin

Since you're on x86 architecture, you're pulling in the official TensorFlow wheels from here: https://pypi.org/project/tensorflow/2.4.1/#history

If you feel motivated or curious about how TensorFlow gets built (it's interesting if you're a compiler nerd, or want to learn about Bazel), this is where I got started: https://github.com/tensorflow/tensorflow/tree/master/tensorflow/tools/ci_build

closed 2022-02-07 by leigh-johnson