bitsy-ai/octoprint-nanny-plugin #92: New plugin bug report

SM-26, 2021-01-25:
- OctoPrint version: 1.5.3
- Print Nanny version: 0.4.0
- OctoPi (or other OS) version: 0.17.0

**Describe the bug**
I can't seem to register my printer

**To Reproduce**
here is the log
``` log
2021-01-25 23:32:37,416 - octoprint.startup - INFO - ******************************************************************************
2021-01-25 23:32:37,418 - octoprint.startup - INFO - Starting OctoPrint 1.5.3
2021-01-25 23:32:37,419 - octoprint.startup - INFO - ******************************************************************************
2021-01-25 23:32:43,921 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: offline
2021-01-25 23:32:43,922 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is not working
2021-01-25 23:32:43,922 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is not working
2021-01-25 23:32:43,931 - octoprint.startup - INFO - We don't appear to be online, not fetching plugin blacklist
2021-01-25 23:32:44,017 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2021-01-25 23:32:45,034 - octoprint.plugin.core - INFO - Plugin Access Anywhere - The Spaghetti Detective (1.4.4) is disabled.
2021-01-25 23:32:56,301 - octoprint.plugin.core - INFO - Plugin LED Strip Control (0.3.7) is disabled.
2021-01-25 23:32:56,315 - octoprint.plugin.core - WARNING - Plugin OctoPrint-IFTTT (1.2.1) is not compatible to Python 3.7.3 (compatibility string: >=2.7,<3).
2021-01-25 23:32:56,779 - octoprint.plugin.core - INFO - Plugin Dashboard (1.14.0) is disabled.
2021-01-25 23:32:57,418 - octoprint.plugin.core - INFO - Found 42 plugin(s) providing 38 mixin implementations, 84 hook handlers
2021-01-25 23:32:57,793 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2021-01-25 23:32:57,823 - octoprint.server - INFO - Intermediary server started
2021-01-25 23:32:57,843 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/pi/.octoprint/uploads...
2021-01-25 23:32:58,184 - octoprint.filemanager.storage - INFO - ... file metadata for /home/pi/.octoprint/uploads initialized successfully.
2021-01-25 23:32:58,188 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2021-01-25 23:32:58,188 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2021-01-25 23:32:58,188 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2021-01-25 23:32:58,188 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2021-01-25 23:32:58,189 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_manage'), Need(method='role', value='plugin_announcements_read')")
2021-01-25 23:32:58,189 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2021-01-25 23:32:58,189 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2021-01-25 23:32:58,190 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2021-01-25 23:32:58,190 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2021-01-25 23:32:58,190 - octoprint.server - INFO - Added new permission from plugin pi_support: PLUGIN_PI_SUPPORT_STATUS (needs: "Need(method='role', value='plugin_pi_support_check')")
2021-01-25 23:32:58,190 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2021-01-25 23:32:58,191 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2021-01-25 23:32:58,191 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2021-01-25 23:32:58,193 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2021-01-25 23:32:58,193 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2021-01-25 23:32:58,211 - octoprint.printer.standard - INFO - Using print time estimator provided by PrintTimeGenius
2021-01-25 23:32:58,977 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python -m pip" as command to invoke pip
2021-01-25 23:32:59,574 - octoprint.util.pip - INFO - Version of pip is 20.2.3
2021-01-25 23:32:59,576 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python3.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2021-01-25 23:32:59,576 - octoprint.util.pip - INFO - ==> pip ok -> yes
2021-01-25 23:32:59,756 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2021-01-25 23:32:59,785 - octoprint.plugins.PrintJobHistory - INFO - Start initializing
2021-01-25 23:32:59,785 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Init DatabaseManager
2021-01-25 23:32:59,786 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Using database in: /home/pi/.octoprint/data/PrintJobHistory/printJobHistory.db
2021-01-25 23:32:59,789 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Check if database-scheme upgrade needed.
2021-01-25 23:32:59,829 - octoprint.plugins.PrintJobHistory.DatabaseManager - INFO - Done DatabaseManager.createDatabase
2021-01-25 23:32:59,830 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Init CameraManager
2021-01-25 23:32:59,830 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Snapshot-Folder:/home/pi/.octoprint/data/PrintJobHistory/snapshots
2021-01-25 23:32:59,830 - octoprint.plugins.PrintJobHistory.CameraManager - INFO - Done CameraMenager
2021-01-25 23:32:59,833 - octoprint.plugins.PrintJobHistory - INFO - Done initializing
2021-01-25 23:32:59,870 - octoprint.plugins.SpoolManager - INFO - Start initializing
2021-01-25 23:32:59,873 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Init DatabaseManager
2021-01-25 23:32:59,874 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Databaseconnection with...
2021-01-25 23:32:59,874 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - {'useExternal': False, 'type': 'sqlite', 'host': 'localhost', 'port': 5432, 'name': 'SpoolDatabase', 'user': 'Olli', 'password': 'illO', 'baseFolder': '/home/pi/.octoprint/data/SpoolManager', 'fileLocation': '/home/pi/.octoprint/data/SpoolManager/spoolmanager.db'}
2021-01-25 23:32:59,878 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database connection succesful. Checking Scheme versions
2021-01-25 23:32:59,878 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Check if database-scheme upgrade needed...
2021-01-25 23:32:59,880 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Current databasescheme: 4
2021-01-25 23:32:59,881 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - ...Database-scheme upgraded not needed.
2021-01-25 23:32:59,881 - octoprint.plugins.SpoolManager.DatabaseManager - INFO - Database created-check done
2021-01-25 23:32:59,882 - octoprint.plugins.SpoolManager - INFO - Done initializing
[... cut ...]
2021-01-25 23:36:35,251 - octoprint.plugins.octoprint_nanny.clients.mqtt - INFO - MQTTClient._on_connect called with client=<paho.mqtt.client.Client object at 0x9e24bbf0> userdata=None rc=4
2021-01-25 23:36:35,252 - octoprint.plugins.octoprint_nanny.clients.mqtt - ERROR - Connection refused by MQTT broker with reason code rc=4
2021-01-25 23:36:35,257 - octoprint.plugins.octoprint_nanny.clients.mqtt - WARNING - Device disconnected from MQTT bridge client=<paho.mqtt.client.Client object at 0x9e24bbf0> userdata=None rc=5

```
**Expected behavior**
it should work? or tell me what I'm doing wrong?

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

CTFishUSA, 2021-01-26:
I've gotten thru registration but also have the rc=4 rc=5 issue.  
Try this from your octoprint server (assuming you are on linux)
openssl s_client -connect mqtt.googleapis.com:8883
openssl s_client -connect mqtt.googleapis.com:443

that's at least one step in proving you can get to the google api through the right port for mqtt.

SM-26, 2021-01-26:
> I've gotten thru registration but also have the rc=4 rc=5 issue.
> Try this from your octoprint server (assuming you are on linux)
> openssl s_client -connect mqtt.googleapis.com:8883
> openssl s_client [...]

the first one is working fine.
the second not
[image]

SM-26, 2021-01-30:
Sorry to say but the problem still persists.

CTFishUSA, 2021-01-30:
If you can't get out to 443, you'll have trouble.  Can you edit your router settings for the pi to unblock port 443 or enable upnp?

SM-26, 2021-01-31:
> If you can't get out to 443, you'll have trouble. Can you edit your router settings for the pi to unblock port 443 or enable upnp?

I don't know what it was, but now both of them are working.
openssl s_client -connect mqtt.googleapis.com:443 and 8883

I haven't changed anything.
just kept nanny up to date, and did a restart to the router over the weekend.

I do really appreciate all of the help.

leigh-johnson, 2021-01-31:
Thank you for following up @SM-26! Please re-open this issue if you notice this again.

I didn't release any fixes that could explain why this suddenly works as of yesterday. 

I'm assuming the root issue is some un-handled/unexpected state in my plugin code, so this might happen to you again! Keep us posted, and thanks again for all of the info provided in this issue. 

[image]

SM-26, 2021-02-01:
I'm sorry if I was unclear.
I still can't register my printer.
I don't know what else to try.

SM-26, 2021-02-05:
With Ver 0.4.3 I was able to register my printer, but it still would not monitor, just stuck at idle.

@leigh-johnson  could you reopen this please?

leigh-johnson, 2021-02-05:
Are you starting monitoring from OctoPrint or Print Nanny's UI? Can you share relevant logs?

Thank you!

leigh-johnson, 2021-05-13:
Thanks for reporting! Just noticed the "Send Wake Command" button is not hooked up. :woman_facepalming: I'll fix this now. I'll also change the button copy so it's more clear what this does - it should send a "start monitoring" command.

In the meantime, does this work for you? 
[image]

leigh-johnson, 2021-05-13:
Here's what you should expect to see when starting monitoring, by the way. :D 

A few things to note:

* You have to refresh the page to see updates to **monitoring active** status. 

This is true of most status indicators in the app right now (Print status, command received/success status). I'm working on making these update without a page refresh!

* The stream should still be visible if you refresh the page, but you might see "web camera is offline" flash or appear for an instant before the stream loads. I'm working on making this more seamless. 

[image]

SM-26, 2021-05-13:
Thank you so much for this detail explanation.

I've sent a print out, and now I'm watching the Nanny watching the printer

Glad I could be a QA assistant, Look mom, I'm (assisting) a developer!
on the same note, is there any chance the Job Status is also not connected?
it's stay at idle
even though the Monitoring Active did update itself.

I did refresh (F5) reload (ctrl+R) and Refresh --nocache (ctrl+F5).
[image]

leigh-johnson, 2021-05-14:
Thanks for pointing this out! I'll dig into why Print Job Status isn't updating.

commit 71df5a1d (2021-01-30): Release v0.4.1 (#98)

* bump print_nanny_client>=0.4.1 ; closes #92 #96

* fix exception occurring when new monitoring settings are applied ; closes #94

* Bump version: 0.4.0 → 0.4.1

* Issue 86 honeycomb traces (#99)

* add spans to honeycomb trace client

* add additional tracing/spans to device registration methods

* add tracing to client and plugin modules

* add more tracing to WorkerManager, PredictorWorker, WebSocketWorker

* fix honeycomb global context initialization

* fix tests

* fix trace context

* factor out relay thread
closed 2022-02-07 by leigh-johnson