bitsy-ai/octoprint-nanny-plugin #93: Webcam Feed Offline

GeoCyberwolf, 2021-01-27:
Taking a snapshot works. but the live feed is offline, sending a wake command doesn't work and print nanny is unable to monitor the printer.

Do I need to port forward?

[image]

leigh-johnson, 2021-01-28:
Thank you for reporting this! Can you open the console and copy/paste all of the messages you see?

The way to do this in Chrome is to open the Chrome Menu in the upper-right-hand corner of the browser window and select More Tools > Developer Tools.

😂 That image looks a little squished too. What's your display resolution?

GeoCyberwolf, 2021-01-28:
`Loading failed for the <script> with source “https://www.googletagmanager.com/gtag/js?id=G-QKHED5DPGV”. login:137:1
Firefox can't establish a connection to the server at wss://print-nanny.com:8080/ws/23/video/download/. 23:432:15
this.$store.state 
Object { socket: Getter & Setter, alerts_dropdown: Getter & Setter, alerts_table: Getter & Setter, settings: Getter & Setter, … }
AlertsDropdown.vue:27:14
Loading failed for the <script> with source “https://www.googletagmanager.com/gtag/js?id=G-QKHED5DPGV”. 23:783:1
Socket connected 
open { target: WebSocket, isTrusted: true, srcElement: WebSocket, eventPhase: 0, bubbles: false, cancelable: false, returnValue: true, defaultPrevented: false, composed: false, timeStamp: 1106, … }
mutations.js:7:12
fetchAlerts 
Object { data: {…}, status: 200, statusText: "OK", headers: {…}, config: {…}, request: XMLHttpRequest }
alerts.js:21:12
setting pagination state 
Object { pageSize: Getter & Setter, totalPages: Getter & Setter, currentPage: Getter & Setter, nextPage: Getter & Setter, previousPage: Getter & Setter, count: Getter & Setter, linkGen: Getter & Setter, … }
mutations.js:11:12
setting data state 
Object { links: Getter & Setter, count: Getter & Setter, current_page: Getter & Setter, page_size: Getter & Setter, total_pages: Getter & Setter, results: Getter & Setter, … }
mutations.js:16:12
this.$store.state 
Object { socket: Getter & Setter, alerts_dropdown: Getter & Setter, alerts_table: Getter & Setter, settings: Getter & Setter, … }
AlertsDropdown.vue:27:14
Object { socket: Getter & Setter, alerts_dropdown: Getter & Setter, alerts_table: Getter & Setter, settings: Getter & Setter, … }
 1 mutations.js:22:12
Socket connected 
open { target: WebSocket, isTrusted: true, srcElement: WebSocket, currentTarget: WebSocket, eventPhase: 2, bubbles: false, cancelable: false, returnValue: true, defaultPrevented: false, composed: false, … }
mutations.js:7:12
Object { socket: Getter & Setter, alerts_dropdown: Getter & Setter, alerts_table: Getter & Setter, settings: Getter & Setter, … }
 1 mutations.js:22:12
Socket connected 
open { target: WebSocket, isTrusted: true, srcElement: WebSocket, currentTarget: WebSocket, eventPhase: 2, bubbles: false, cancelable: false, returnValue: true, defaultPrevented: false, composed: false, … }`

I think the issue is with "Firefox can't establish a connection to the server at wss://print-nanny.com:8080/ws/23/video/download/."

My display is 1920x1080

closed 2022-02-07 by leigh-johnson