bitsy-ai/octoprint-nanny-plugin #95: https://print-nanny.com/dashboard/octoprint-devices/ page has too wide right side frame

CTFishUSA, 2021-01-27:
- OctoPrint version: Website https://print-nanny.com/dashboard/octoprint-devices/
- Print Nanny version: n/a
- OctoPi (or other OS) version: n/a

**Describe the bug**
Rightmost column is off the page.  Like the frame is not scaling properly for my screen.  If this is a code issue on your end, suggest to fix.  If on my end (Raspberry 4 with Chromium), close with no change.

**To Reproduce**
Steps to reproduce the behavior:
1. Visit https://print-nanny.com/dashboard/octoprint-devices/
2. See error

**Expected behavior**
The page should fit all of my information (RAM is cut off).  Is it because my OctoPrint Device name is too long?  Can you add a scrollbar?

[image]

**Screenshots**
If applicable, add screenshots to help explain your problem.

leigh-johnson, 2021-01-28:
Thank you! I'll add some overflow or word wrap

[image]

leigh-johnson, 2021-01-31:
I ended up sticking most of these fields behind a "view metadata" button. You'll see this update in release 0.4.2, which I'll roll out in an hour or three. Thanks again for reporting this! Closing this issue for now. 

[image]

Actual footage of me adjusting the CSS and then giving up

https://media1.tenor.com/images/614c9b4639a2588383f47e138177da81/tenor.gif?itemid=12014506

closed 2021-01-31 by leigh-johnson