bitsy-ai/octoprint-nanny-plugin #96: can't register my printer

PGlyre, 2021-01-27:
- OctoPrint version: 1.5.3
- Print Nanny version: 0.4.0
- OctoPi (or other OS) version: 0.17.0

**Describe the bug**
I have installed Nanny with you script and first install ATLAS as you explain and all seems to 
work except from the part to register my Printer. I can see my printer on the web interface but in Idle mode. All information about my RPi is correct so Nanny have got correct information in somehow

**Screenshots**
[image]

[octoprint1.log](https://github.com/bitsy-ai/octoprint-nanny-plugin/files/5878528/octoprint1.log)

leigh-johnson, 2021-01-28:
🙇‍♀️ Thank you for reporting this! Looks like validation on this printer profile field is a bit too aggressive for the real-world. 

```
2021-01-27 08:54:36,712 - octoprint.plugins.octoprint_nanny.manager - WARNING - Discarding plugin_octoprint_nanny_device_register_failed with payload {'event_type': 'plugin_octoprint_nanny_device_register_failed', 'event_data': {'msg': '{"volume_custom_box":["Must be a valid boolean."]}'}}
```

PGlyre, 2021-01-28:
Hi, I have to ask because I do not understand. If I have any aggressive configurations, it is probably due to low knowledge on my part. I have a Prusa MK3S and use a very basic configuration, I thought ....... Do you have any idea what kind of fault I have or it has nothing to do with Nanny at all?

leigh-johnson, 2021-01-28:
@PGlyre I apologise for the ambiguity! You don't need to do anything. 

I'll fix this by permitting null values for this field in the next bug release. 

Thanks again for the report!

leigh-johnson, 2021-01-30:
Hey @PGlyre, thanks again for opening a bug report. I've released a fix for in 0.4.1. Please re-open if this doesn't resolve the issue! :bowing_woman: 

https://github.com/bitsy-ai/octoprint-nanny-plugin/archive/v0.4.1.zip

PGlyre, 2021-01-30:
Hi, i have now tested after upgrade ti 0.4.1 but i'm sorry to say that i still not work to register my Printer. I have try to uninstall and reinstall i't again but no
[octoprint (1).log](https://github.com/bitsy-ai/octoprint-nanny-plugin/files/5897642/octoprint.1.log)

leigh-johnson, 2021-01-30:
Thanks for the logs. :bowing_woman: 

Ah-hah, after digging into the OctoPrint code, I can see this field is duck-typed to be `False` or a dict. I'll update the back-end model accordingly. https://github.com/OctoPrint/OctoPrint/blob/f67c15a9a47794a68be9aed4f2d5a12a87e70179/src/octoprint/printer/profile.py#L46

```
print_nanny_client.exceptions.ApiException: (400)
Reason: Bad Request
HTTP response headers: <CIMultiDictProxy('Content-Type': 'application/json', 'Vary': 'Accept, Accept-Language, Cookie', 'Allow': 'POST, OPTIONS', 'X-Frame-Options': 'DENY', 'Content-Length': '50', 'Content-Language': 'en', 'Strict-Transport-Security': 'max-age=60; includeSubDomains; preload', 'X-Content-Type-Options': 'nosniff', 'X-XSS-Protection': '1; mode=block', 'Referrer-Policy': 'same-origin', 'Date': 'Sat, 30 Jan 2021 07:32:35 GMT', 'Via': '1.1 google', 'Alt-Svc': 'clear')>
HTTP response body: {"volume_custom_box":["Must be a valid boolean."]}
```

commit b0c74af2 (2021-02-01): Release v0.4.2 (#101)

* use OctoPrint Events enum in WorkerManager_remote_control_event_handlers

* update octoprint device.monitoring_active

* bump print_nanny_client version; closes #96

* Issue 92 configurable mqtt port (#104)

* delete unused tempalte

* refactor settings memoization into mixin for sanity

* freeze before adding pyre typechecker

* refactor rest and mqtt client initialization to raise PluginSettingsRequired

* add passing test test_telemetry_queue_send_loop_valid_octoprint_event

* add passing test for test_telemetry_queue_send_loop_bounding_box_predict

* add passing te [... cut]
closed 2021-02-01 by leigh-johnson