bitsy-ai/octoprint-nanny-plugin #105: Error! Something went wrong while re-provisioning this device.

dyoga, 2021-01-31:
[image]

Hello i can't register my printer

Printer
Creality CR-6 SE

Octopi Ver.
1.5.3

Octoprint Nanny Ver.
0.4.1

leigh-johnson, 2021-02-01:
Hey @dyoga, thanks for the report!

Can you try this again with 0.4.2 and let me know how it goes? 

If that doesn't work for you, attach the logs in ~/.octoprint/logs so I can look into this. Thanks again!

dyoga, 2021-02-01:
Hello, now the connection was succsesfull.
But my Printer is shown as "idle" with active monitoring :(

> Hey @dyoga, thanks for the report!
> 
> Can you try this again with 0.4.2 and let me know how it goes?
> 
> If that doesn't work for you, please attach the logs in ~/.octoprint/logs so I can look into this. Thanks again!

leigh-johnson, 2021-02-01:
Woo glad device registration worked for you!

Bummer on the "idle" status, I must have missed a edge case. Can you collect the following info to help me reproduce?

1. Does this happen when you start monitoring in Print Nanny's UI, OctoPrint's UI, or both?
2. Can you start monitoring, sing a short song like "baby shark", stop monitoring, and then upload ~/.octoprint/logs for me to inspect?

Thanks!

dyoga, 2021-02-02:
Hello, you are very fast :D

Monitoring starts now but no livestream. I'm using Octolapse if it's relevant.

There is no Octoprint Nanny log. which log should i upload? The Octoprint log is about 490MB :o

[image]

> Woo glad device registration worked for you!
>
> Bummer on the "idle" status, I must have missed a edge case. Can you collect the following info to help me reproduce?
>
> 1. Does this happen when you [...]

leigh-johnson, 2021-02-02:
Oh, thank you for pointing out the log issue! 

I just realized OctoPrint Nanny is logging to octoprint.log, probably because I mucked around with the root logger for multi-process log aggregation in an earlier release.

I'll fix this up so OctoPrint Nanny follows the convention of logging to its own file.

Side note, I love how much functionality Gina has built into OctoPrint. Setting up log handlers automatically is such a great decision. 😍

closed 2022-02-07 by leigh-johnson