bitsy-ai/octoprint-nanny-plugin #112: abilty to skip first run wizard.

DrDrache, 2021-02-03:
**Describe your Request**
The First run wizard came up after installing the plugin, it is unskippable for me. so my octoprint instance is unusable because I am not accepted into the beta as of yet.
For this reason, OR in instances where the plugin is installed but the user is not interested in an immediate configuration (lack of time; lack of account; lack of desire; other priorities), I suggest this window be actually skippable

the wording "Unless otherwise noted, you may just skip any wizard page by clicking "Next" or "Finish"." does appear, but there is no way to skip the first page.

leigh-johnson, 2021-02-06:
Hey @DrDrache, thanks for writing in about this. Can you describe what happens when you try to skip the Print Nanny wizard? A video or gif would be helpful. I used peek to record the gif below. https://github.com/phw/peek

I tried to reproduce this on a fresh OctoPrint installation with 0.4.3 installed. I was able to finish the wizard without touching Print Nanny's settings. Are you seeing a different behavior?

[image]

DrDrache, 2021-02-06:
here is a GIF of the behavior - even after a power cycle, I seem to be stuck.
I am not in immediate need of this instance, so I left it for troubleshooting.. 

maybe there is a conflict with one of my other plugins installed such as :
https://github.com/LazeMSS/OctoPrint-UICustomizer

[image]

leigh-johnson, 2021-02-07:
Thanks for the deets @DrDrache! If you're feeling up for it, can you disable OctoPrint-UICustomizer and let me know if you're able to dismiss Print Nanny's wizard afterwards?

To get back into a usable state, you can uninstall the plugin by sshing into your Pi and running: 

```
/home/pi/oprint/bin/pip uninstall octoprint-nanny
```

Note: the path above only applies to the OctoPi image. If you installed OctoPrint yourself, I trust you'll be able to find the install location and perform similar steps. Good luck!

DrDrache, 2021-02-17:
I was able to remove the plugin, but yet as of now, I cannot find which plugin is causing my issues.

in addition - whatever issue I am having causes the octopi to also lock up and require a power cycle if left for more than a few min.
I haven't had time to go through every plugin yet... I have attached a list of my plugins in case someone reading can tell me which is the most likely suspect.

Arc-Welder (1.0.0+u.bb71e8f) 
Autologin Configuration (0.2.0)
Autoscroll (0.0.3) 
Bed Visualizer (1.0.1rc2) 
BetterHeaterTimeout (1.3.0) 
Change Filament Plugin (0.3.2) 
Dashboard (1.17.0) < -- maybe?
Detailed Progress (0.2.7) 
DisplayLayerProgress Plugin (1.25.4) 
Exclude Region (0.3.0) H
Firmware Updater (1.8.1rc1) 
Marlin EEPROM Editor (3.0.1) 
Marlin GCode Documentation (0.12.0) 
Multi Line Terminal (0.1.5) 
Navbar Temperature Plugin (0.14) <--- maybe
Octolapse (0.4.1) 
OctoPrint-WiFiStatus (1.5.1) 
PrettyGCode (1.2.4) 
Printer Stats (3.1.2) 
PrintTimeGenius Plugin (2.2.7) 
Resource Monitor (0.3.1) 
Simple Emergency Stop (1.0.4) 
SpoolManager Plugin (1.3.3) 
Terminal Messaging (0.1.0) 
Themeify (1.2.2) <-- also maybe?
TP-Link Smartplug (1.0.0rc1)
Webcam Iframe (0.0.5) 
Z Probe Offset (0.5)

DrDrache, 2021-04-01:
just want to update -
I have systematically disabled any plugin that has effects for the UI - sadly - during this; the Print-nanny plugin stopped installing/uninstalling properly.

`pi@octopi:~ $ /home/pi/oprint/bin/pip uninstall octoprint-nanny`
`Found existing installation: OctoPrint-Nanny 0.4.3`
`Uninstalling OctoPrint-Nanny-0.4.3:`
  `Would remove:`
  `  /home/pi/oprint/lib/python3.7/site-packages/OctoPrint_Nanny-0.4.3.dist-info/*`
   ` /home/pi/oprint/lib/python3.7/site-packages/octoprint_nanny/*`
`Proceed (y/n)? y`
  `Successfully uninstalled OctoPrint-Nanny-0.4.3`
`pi@octopi:~ $`

which then; upon install of the plugin, it is missing from the plug-in list; meaning any other testing is void.
I, unfortunately, am not sure of why it doesn't work - I will debug this if I am pointing in the right direction, but I'm at the limits of my knowledge of this part of the stack.

leigh-johnson, 2021-05-10:
Hey @DrDrache, I'm going to close this issue since there hasn't been activity here. Can you try v0.6.0 of the plugin and open a new issue if you're still unable to use Print Nanny? Thanks!

closed 2021-05-10 by leigh-johnson