bitsy-ai/octoprint-nanny-plugin #116: Receive TFLite model version via device configuration topic

leigh-johnson, 2021-02-06:
**Describe your Request**
Currently, the model weights are distributed via package data. This makes plugin updates (source distribution from .zip) take much longer than necessary. 

Instead, the device should download the latest model to the plugin data directory. Model version updates will be published via configuration topic: https://cloud.google.com/iot/docs/concepts/devices#device_configuration

** Benchmarks **

0.4.3
```
time pip install --no-cache-dir --force-reinstall https://github.com/bitsy-ai/octoprint-nanny-plugin/archive/v0.4.3.zip
```

CTFishUSA, 2021-02-07:
Is this the end of the 120mb wheel?

leigh-johnson, 2021-02-07:
> Is this the end of the 120mb wheel?

Yep =) I was using the full TensorFlow binary for some extra profiling stuff that only lives in the full package. This release will switch over to using TensorFlow's runtime package, which has a much smaller footprint.

More info here:
https://www.tensorflow.org/lite/guide/python#about_the_tensorflow_lite_runtime_package

leigh-johnson, 2021-04-10:
Punting this out to release 0.6.0 for now.

closed 2021-05-07 by leigh-johnson