bitsy-ai/octoprint-nanny-plugin #120: Verify key pair download against checksum

leigh-johnson, 2021-02-09:
**Describe your Request**
Great suggestion from @CTFishUSA! Send a checksum with key pair and verify saved files match checksum.

CTFishUSA, 2021-02-09:
Take the private key.
Run the same openssl command from your serger side code, but execute on the local octoprint instance instead of the server side.  Assuming openssl is available.  (Does that need to be a requirement now?).  Outputs the public key.

Run the same hash on the public key, locally, to output the fingerprint.  

Compare the fingerprint and generate a "good" or "bad" match message for your log file.

leigh-johnson, 2021-02-09:
I thought about this for a hot minute, and I think a simple checksum of the contents will actually be more reliable!

Check out this nifty diagram depicting the contents of an RSA private key. If an integrity check fails on the contents of n, e, d (red, purple, green), I believe the key will still produce a cryptographic signature that can still be decoded with the key's public side.  
https://crypto.stackexchange.com/questions/31807/why-does-my-ssh-private-key-still-work-after-changing-some-bytes-in-the-file

Your intermittent success/failure presents as a network issue ... but another solid theory is that you're occasionally hitting a bad block on your SD card when saving key files. Comparing the contents written to disk against a checksum catch that; cryptographic signing and public key extraction rely on SOME parts of the key being intact. 

You can also look for bad blocks by running:
```
sudo badblocks -sv /dev/mmcblk0p2
```

closed 2021-03-06 by leigh-johnson